openai/mujoco-py #710: RuntimeError: Failed to initialize OpenGL

DexiongYung, 2022-05-06:
**Describe the bug**

```
Found 1 GPUs for rendering. Using device 0.
Traceback (most recent call last):
  File "/coc/pskynet4/dyung6/CS7000/train.py", line 189, in <module>
    main(cfg)
  File "/coc/pskynet4/dyung6/CS7000/train.py", line 73, in main
    obs = envs.reset()
  File "/coc/pskynet4/dyung6/CS7000/env.py", line 81, in reset
    obs = self.venv.reset()
  File "/coc/pskynet4/dyung6/CS7000/env.py", line 31, in reset
    obs = self.venv.reset()
  File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py", line 61, in reset
    obs = self.envs[env_idx].reset()
  File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/stable_baselines3/common/monitor.py", line 79, in reset
    return self.env.reset(**kwargs)
  File "/coc/pskynet4/dyung6/CS7000/gyms/SB3gym.py", line 29, in reset
    rgb_obs = self.env.render(
  File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/gym/core.py", line 295, in render
    return self.env.render(mode, **kwargs)
  File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/gym/envs/mujoco/mujoco_env.py", line 157, in render
    self._get_viewer(mode).render(width, height, camera_id=camera_id)
  File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/gym/envs/mujoco/mujoco_env.py", line 186, in _get_viewer
    self.viewer = mujoco_py.MjRenderContextOffscreen(self.sim, -1)
  File "mjrendercontext.pyx", line 46, in mujoco_py.cymj.MjRenderContext.__init__
  File "mjrendercontext.pyx", line 114, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
```

**To Reproduce**
**.bashrc**
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/nethome/dyung6/.mujoco/mujoco210/bin
export LD_PRELOAD=/nethome/dyung6/share4_dyung6/anaconda3/envs/7000/lib/libGLEW.so
# export LD_PRELOAD=/nethome/dyung6/share4_dyung6/anaconda3/envs/7000/lib/libGLEW.so:/usr/lib/nvidia-460/libGL.so
```

This issue seems to arise when you all render() I followed this link solution [https://github.com/openai/mujoco-py/issues/187](url), which is the uncommented `export LD_PRELOAD=/nethome/dyung6/share4_dyung6/anaconda3/envs/7000/lib/libGLEW.so:/usr/lib/nvidia-460/libGL.so`. Which didn't solve the problem. The version above is what I have tried most recently.

**Expected behavior**
Should not crash

**Error Messages**
Including more/longer error messages gives us more information to work with.

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04.6 LTS
 - Python Version 3.10.4
 - Mujoco Version 210
 - mujoco-py version 2.1.2.14

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 :/usr/lib/nvidia:/nethome/dyung6/.mujoco/mujoco210/bin
 - output of: `echo $HOME`
 /nethome/dyung6
 - output of: `echo $USER`
 dyung6

**Additional context**
Add any other context about the problem here.

mehranraisi, 2022-08-03:
It seems Conda has its internal GCC library. I had searched for hours, and the only thing that worked for me was to uninstall these  libraries:
conda uninstall _libgcc_mutex
conda uninstall glfw

mehranraisi, 2022-08-03:
try :
conda uninstall gcc
conda install -c conda-forge gcc==12.1.0

It worked for me!

Logan-lxw, 2024-03-28:
Have you resolved your problem?I meet the same one.I need your help,thank you!

> **Describe the bug**
> 
> ```
> Found 1 GPUs for rendering. Using device 0.
> Traceback (most recent call last):
>   File "/coc/pskynet4/dyung6/CS7000/train.py", line 189, in <module>
>     main(cfg)
>   File "/coc/pskynet4/dyung6/CS7000/train.py", line 73, in main
>     obs = envs.reset()
>   File "/coc/pskynet4/dyung6/CS7000/env.py", line 81, in reset
>     obs = self.venv.reset()
>   File "/coc/pskynet4/dyung6/CS7000/env.py", line 31, in reset
>     obs = self.venv.reset()
>   File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py", line 61, in reset
>     obs = self.envs[env_idx].reset()
>   File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/stable_baselines3/common/monitor.py", line 79, in reset
>     return self.env.reset(**kwargs)
>   File "/coc/pskynet4/dyung6/CS7000/gyms/SB3gym.py", line 29, in reset
>     rgb_obs = self.env.render(
>   File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/gym/core.py", line 295, in render
>     return self.env.render(mode, **kwargs)
>   File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/gym/envs/mujoco/mujoco_env.py", line 157, in render
>     self._get_viewer(mode).render(width, height, camera_id=camera_id)
>   File "/srv/kira-lab/share4/dyung6/anaconda3/envs/7000/lib/python3.10/site-packages/gym/envs/mujoco/mujoco_env.py", line 186, in _get_viewer
>     self.viewer = mujoco_py.MjRenderContextOffscreen(self.sim, -1)
>   File "mjrendercontext.pyx", line 46, in mujoco_py.cymj.MjRenderContext.__init__
>   File "mjrendercontext.pyx", line 114, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
>   File "opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
> RuntimeError: Failed to initialize OpenGL
> ```
> 
> **To Reproduce** **.bashrc**
> 
> ```
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/nethome/dyung6/.mujoco/mujoco210/bin
> export LD_PRELOAD=/nethome/dyung6/share4_dyung6/anaconda3/envs/7000/lib/libGLEW.so
> # export LD_PRELOAD=/nethome/dyung6/share4_dyung6/anaconda3/envs/7000/lib/libGLEW.so:/usr/lib/nvidia-460/libGL.so
> ```
> 
> This issue seems to arise when you call `render()` I followed this link's solution [https://github.com/openai/mujoco-py/issues/187](url), which is the uncommented `export LD_PRELOAD=/nethome/dyung6/share4_dyung6/anaconda3/envs/7000/lib/libGLEW.so:/usr/lib/nvidia-460/libGL.so`. Which didn't solve the problem. The version above is what I have tried most recently. Both give the same error.
> 
> **Expected behavior** Should not crash
> 
> **Error Messages** Including more/longer error messages gives us more information to work with.
> 
> **Desktop (please complete the following information):**
> 
> * OS: Ubuntu 16.04.6 LTS
> * Python Version 3.10.4
> * Mujoco Version 210
> * mujoco-py version 2.1.2.14
> 
> **Environment**
> 
> * output of: `echo $LD_LIBRARY_PATH`
>   :/usr/lib/nvidia:/nethome/dyung6/.mujoco/mujoco210/bin
> * output of: `echo $HOME`
>   /nethome/dyung6
> * output of: `echo $USER`
>   dyung6

Logan-lxw, 2024-03-28:
@DexiongYung