openai/mujoco-py #714: RuntimeError: Failed to initialize OpenGL

QUIlToT, 2022-05-12:
I know this type of problem has been asked many times, and I have tried to follow these methods (#665, #187) to solve my problem, but it seems useless. 

Here is the code:

```python
import mujoco_py

MODEL_XML = """
<?xml version="1.0" ?>
<mujoco>
    <worldbody>
    <camera name="cam0" pos="-0.1 0.1 0.0" quat="0.707 0.707 0 0" />
        <body name="box" pos="0 0 0.2">
            <geom size="0.15 0.15 0.15" type="box"/>
            <joint axis="1 0 0" name="box:x" type="slide"/>
            <joint axis="0 1 0" name="box:y" type="slide"/>
        </body>
        <body name="floor" pos="0 0 0.025">
            <geom size="1.0 1.0 0.02" rgba="0 1 0 1" type="box"/>
        </body>
    </worldbody>
</mujoco>
"""
model = mujoco_py.load_model_from_xml(MODEL_XML)
sim = mujoco_py.MjSim(model)
viewer = mujoco_py.MjViewer(sim)
image = sim.render(width='300', height = '300', camera_name='cam0', depth=False)
print(image)

```

Then I met the following errors:

File "mujoco_py/mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
File "mujoco_py/mjsim.pyx", line 158, in mujoco_py.cymj.MjSim.render
File "mujoco_py/mjrendercontext.pyx", line 45, in mujoco_py.cymj.MjRenderContext.__init__
File "mujoco_py/mjrendercontext.pyx", line 113, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
File "mujoco_py/opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL

Linux system: Ubuntu 18.04.
Mujoco_py version: 2.0.2.5.

It seems that commenting  `LD_PRELOAD ` is useless for me.  I have no libGL.so in `/usr/lib/nvidia-510`.

Has anyone successfully solved this problem? I would appreciate a response! Thx!

closed 2022-05-12 by QUIlToT

Hazmatius, 2022-06-01:
Did you close this because you solved the issue? What did you end up doing?

DDCoan, 2022-06-24:
Hi, how did you solve this issue? Could you pls maybe share your solution? many thanks!

QUIlToT, 2022-06-28:
Hi, I'm sorry for replying now. I can't remember the exact details, but I can list all the methods I've tried below. 

(1) [https://github.com/openai/mujoco-py/issues/187](url)

(2) Check the `device_id` in the programming. Some related issues can be found. 

Hope these are helpful to you!

Irismoon, 2022-07-11:
> many thanks!

Hi, have you addressed the issue?

Thanks!

DDCoan, 2022-07-12:
`unset LD_PRELOAD` worked for me. did you try that?

Dimoo-POP, 2022-07-12:
> `unset LD_PRELOAD` worked for me. did you try that?

Hi! Should this command (`unset LD_PRELOAD`) be entered in the terminal, before starting the IDE?

DDCoan, 2022-07-12:
you can enter `unset LD_PRELOAD` in terminal then "python `yourfile.py` to run your code

Dimoo-POP, 2022-07-13:
> you can enter `unset LD_PRELOAD` in terminal then "python `yourfile.py` to run your code

Thank you!

Irismoon, 2022-07-25:
> `unset LD_PRELOAD` worked for me. did you try that?

Yes, I tried that. But that did not help with the error. And finally this problem is addressed by using GPU in my case.

irfanrah, 2022-07-29:
> unset LD_PRELOAD worked for me. did you try that?

Hi, I tried this and got this error

> Creating window glfw
> ERROR: GLEW initalization error: Missing GL version
> Press Enter to exit

xuanxuanzzzii, 2024-11-14:
> > `unset LD_PRELOAD` worked for me. did you try that?`unset LD_PRELOAD` 对我有用。你试过吗？
> 
> Yes, I tried that. But that did not help with the error. And finally this problem is addressed by using GPU in my case.是的，我试过了。但这对错误没有帮助。最后，在我的情况下，通过使用 GPU 解决了这个问题。

hello, how did you finally solve this problem?