openai/mujoco-py #716: Confusion about the number of GPU

QUIlToT, 2022-05-14:
When I tried to render the Mujoco model with a specified GPU, I got some confusion, which may be very simple. 

When the following code is executed, I can find ``` Found 3 GPUs for rendering. Using device 0 ```  in the Pycharm console window.

``` python
image = sim.render(width=300, height=300, camera_name = 'cam0', depth=False, device_id=0)
```

However, when I use pytorch to detect the number and name of GPU, the related codes and displays are shown below
``` python
print('Number of gpu: ', torch.cuda.device_count())
print('its name： ', torch.cuda.get_device_name(0))

Number of gpu: 1
its name：  NVIDIA GeForce RTX 3070 Ti
```

Why the number of GPUs detected by the two methods is different?

For Mujoco, how can I check the name of the detected GPUs?

Hope to get some replies!

kano00, 2022-07-18:
Hi. I have the same problem. Have you found the answer?

QUIlToT, 2022-07-19:
> Hi. I have the same problem. Have you found the answer?

Sry, no idea right now...

BetterZH, 2022-11-07:
I have the same problem. Have you found the answer?