openai/mujoco-py #737: Illegal Instruction (core dumped)

meeww, 2022-08-01:
Following along with https://github.com/openai/mujoco-py on Windows 10 running a Ubuntu (64-bit) VM I run into the error:

Illegal Instruction (core dumped)

It happens after running "model = mujoco_py.load_model_from_path(xml_path)" 
I have also tried following along with another guide in which I got the same error message from running the "./simulate ../model/humanoid.xml" command inside the bin directory.

Online I read something about it possibly being related to my CPU being too old? I have an Intel i7-4930k which I think should be capable, but I'm not sure.

If anyone knows what I'm doing wrong I would greatly appreciate some help! :) 
Thanks

saran-t, 2022-09-08:
This seems to be a `mujoco` question rather than `mujoco-py`. If you repost the issue on https://github.com/deepmind/mujoco we could try to help. (When you do create the issue over there, I'll need to know how you've set up your VM, and the output of `cat /proc/cpuinfo` on your VM).

lzhhh93, 2023-05-04:
Hi @meeww, did you get the problem solved? Thanks for some instructions.