openai/mujoco-py #738: CompileError: command 'gcc' failed with exit status 1

newne, 2022-08-03:
**Describe the bug**
I aleady install the `mujoco` :
[image]

but when i install mujoco_py, some error happened and have no solution to solve it yet. i install with these command :
```
cd ~/mujoco-py
pip install -r requirements.txt
pip install -r requirements.dev.txt
python setup.py install
```
everything is ok ,but can not use it in python. it always came out this error.
**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**
```
(trajectory) ncg@ncg-pc:~/Downloads/mujoco-py$ ipython
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.3 -- An enhanced Interactive Python. Type '?' for help.

In [1]: 

In [1]: import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/ncg/anaconda3/envs/trajectory/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/ncg/Downloads/mujoco-py/mujoco_py -I/home/ncg/.mujoco/mujoco210/include -I/home/ncg/anaconda3/envs/trajectory/lib/python3.6/site-packages/numpy/core/include -I/home/ncg/Downloads/mujoco-py/mujoco_py/vendor/egl -I/home/ncg/anaconda3/envs/trajectory/include/python3.6m -c /home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c -o /home/ncg/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_2.1.2.14_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c: In function ‘__pyx_f_9mujoco_py_4cymj_8PyMjData__set’:
/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c:61635:78: error: ‘mjData’ {aka ‘struct mjData_’} has no member named ‘qfrc_unc’
61635 |   __pyx_t_4 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_mjtNum_1d(__pyx_v_p->qfrc_unc, __pyx_t_3)); if (unlikely(!__pyx_t_4)) __PYX_ERR(4, 2943, __pyx_L1_error)
      |                                                                              ^~
/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c:61654:78: error: ‘mjData’ {aka ‘struct mjData_’} has no member named ‘qacc_unc’
61654 |   __pyx_t_4 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_mjtNum_1d(__pyx_v_p->qacc_unc, __pyx_t_3)); if (unlikely(!__pyx_t_4)) __PYX_ERR(4, 2944, __pyx_L1_error)
      |                                                                              ^~
/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c: In function ‘__pyx_pf_9mujoco_py_4cymj_12PyMjrContext_15glewInitialized___get__’:
/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c:84600:55: error: ‘mjrContext’ {aka ‘struct mjrContext_’} has no member named ‘glewInitialized’; did you mean ‘glInitialized’?
84600 |   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->glewInitialized); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 4175, __pyx_L1_error)
      |                                                       ^~~~~~~~~~~~~~~
      |                                                       glInitialized
/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c: In function ‘__pyx_pf_9mujoco_py_4cymj_12PyMjrContext_15glewInitialized_2__set__’:
/home/ncg/Downloads/mujoco-py/mujoco_py/cymj.c:84655:22: error: ‘mjrContext’ {aka ‘struct mjrContext_’} has no member named ‘glewInitialized’; did you mean ‘glInitialized’?
84655 |   __pyx_v_self->ptr->glewInitialized = __pyx_v_x;
      |                      ^~~~~~~~~~~~~~~
      |                      glInitialized
---------------------------------------------------------------------------
DistutilsExecError                        Traceback (most recent call last)
~/anaconda3/envs/trajectory/lib/python3.6/distutils/unixccompiler.py in _compile(self, obj, src, ext, cc_args, extra_postargs, pp_opts)
    117             self.spawn(compiler_so + cc_args + [src, '-o', obj] +
--> 118                        extra_postargs)
    119         except DistutilsExecError as msg:

~/anaconda3/envs/trajectory/lib/python3.6/distutils/ccompiler.py in spawn(self, cmd)
    908     def spawn(self, cmd):
--> 909         spawn(cmd, dry_run=self.dry_run)
    910 

~/anaconda3/envs/trajectory/lib/python3.6/distutils/spawn.py in spawn(cmd, search_path, verbose, dry_run)
     35     if os.name == 'posix':
---> 36         _spawn_posix(cmd, search_path, dry_run=dry_run)
     37     elif os.name == 'nt':

~/anaconda3/envs/trajectory/lib/python3.6/distutils/spawn.py in _spawn_posix(cmd, search_path, verbose, dry_run)
    158                           "command %r failed with exit status %d"
--> 159                           % (cmd, exit_status))
    160             elif os.WIFSTOPPED(status):

[... 126 lines cut ...]

CompileError: command 'gcc' failed with exit status 1
```

**Desktop (please complete the following information):**
 - OS: ubuntu 22.04
 - Python Version 3.6
 - Mujoco Version 210(lastest)
 - mujoco-py version 2.1

**Environment**
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ncg/.mujoco/mujoco210/bin:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ncg/.mujoco/mujoco210/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ncg/.mujoco/mujoco210/include/mujoco:$LD_LIBRARY_PATH
#export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ncg/.mujoco/mujoco210/include:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ncg/.mujoco/mujoco210/bin
# cuda
export PATH=/usr/local/cuda-11.1/bin:$PATH
export LD_LIBRARY_PATH=/usr/local/cuda-11.1/lib64:$LD_LIBRARY_PATH
```

**Additional context**
Add any other context about the problem here.

CT702-7, 2022-09-01:
Same problem as I met, but that was just one of wired problems...

saran-t, 2022-09-05:
If you're looking start using MuJoCo in a new project, please use https://github.com/deepmind/mujoco (aka `pip install mujoco`) instead.

`mujoco-py` is no longer being developed or maintained.