openai/mujoco-py #744: Enter bug when try to install mujoco

qinmoelei, 2022-10-09:
My python version is 3.8.5.
I download the MuJoCo version 2.1 binaries for [Linux](https://mujoco.org/download/mujoco210-linux-x86_64.tar.gz) and extract to `~/.mujoco/mujoco210`.
I use the following code to test the installation.
```
import os

os.environ[ 'LD_LIBRARY_PATH'] = '$LD_LIBRARY_PATH:/home/sunshuo/.mujoco/mujoco210/bin'
import mujoco_py
print(os.environ.get("LD_LIBRARY_PATH", ""))
mj_path = mujoco_py.utils.discover_mujoco()
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
model = mujoco_py.load_model_from_path(xml_path)
sim = mujoco_py.MjSim(model)
print(sim.data.qpos)
```
And it shows the exception
```
Traceback (most recent call last):
  File "/home/sunshuo/qml/decision-transformer/gym/test.py", line 8, in <module>
    import mujoco_py
  File "/home/sunshuo/miniconda3/envs/decision-transformer-gym/lib/python3.8/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/sunshuo/miniconda3/envs/decision-transformer-gym/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/sunshuo/miniconda3/envs/decision-transformer-gym/lib/python3.8/site-packages/mujoco_py/builder.py", line 76, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", get_nvidia_lib_dir())
  File "/home/sunshuo/miniconda3/envs/decision-transformer-gym/lib/python3.8/site-packages/mujoco_py/builder.py", line 120, in _ensure_set_env_var
    raise Exception("\nMissing path to your environment variable. \n"
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sunshuo/.mujoco/mujoco210/bin
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia
```
But if I add the `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia`, then it shows that I should add `$LD_LIBRARY_PATH:/home/sunshuo/.mujoco/mujoco210/bin` instead.

fionalluo, 2024-03-06:
I have the same bug, how was it resolved?

meghbhalerao, 2024-10-08:
same issue ... has anyone fixed it? My error is the following - 
```
Traceback (most recent call last):
  File "human_robot_gym/demos/demo_reach_human_environment.py", line 12, in <module>
    import robosuite as suite
  File "/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/robosuite/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/robosuite/environments/base.py", line 4, in <module>
    from mujoco_py import MjRenderContextOffscreen, MjSim, load_model_from_xml
  File "/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/mujoco_py/__init__.py", line 15, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/mujoco_py/builder.py", line 499, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/mujoco_py/builder.py", line 71, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", get_nvidia_lib_dir())
  File "/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/mujoco_py/builder.py", line 115, in _ensure_set_env_var
    raise Exception("\nMissing path to your environment variable. \n"
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=/home/mb230/anaconda3/envs/hrgym/lib/python3.8/site-packages/mujoco_py/binaries/linux/mujoco210/bin
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-560
```
Even though my `LD_LIBRARY_PATH` already has the correct values when I do `echo $LD_LIBRARY_PATH` like this  
```
echo $LD_LIBRARY_PATH
/usr/local/cuda-12.6/lib64:/home/mb230/.mujoco/mujoco210/bin:/usr/lib/nvidia:/usr/lib/nvidia-560
```
and also - 
```
In [4]: os.getenv("LD_LIBRARY_PATH")
Out[4]: '/usr/local/cuda-12.6/lib64:/home/mb230/.mujoco/mujoco210/bin:/usr/lib/nvidia:/usr/lib/nvidia-560'

```
so looks like my `LD_LIBRARY_PATH` has correct values, but it is not being identified by mujoco, can someone please help? thanks!