openai/mujoco-py #745: fatal error: GL/glew.h: No such file or directory     4 | #include <GL/glew.h>

qinmoelei, 2022-10-10:
My python version is 3.8.5
My problem is that after I have installed the mujoco and try it using the following code
```
import os
import mujoco_py
print(os.environ.get("LD_LIBRARY_PATH", ""))
mj_path = mujoco_py.utils.discover_mujoco()
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
model = mujoco_py.load_model_from_path(xml_path)
sim = mujoco_py.MjSim(model)

print(sim.data.qpos)
```
The following error occurs
```
 fatal error: GL/glew.h: No such file or directory
    4 | #include <GL/glew.h>
      |          ^~~~~~~~~~~
compilation terminated.

```
The problem is that I do not have the root authority to use the command 
```
sudo apt-get install libglew-dev

```
What should I do?

IpsumDominum, 2022-10-19:
umm do `sudo apt-get install libglew-dev`?

Kelym, 2023-01-15:
You can follow [#627 ](https://github.com/openai/mujoco-py/issues/627)

238Ver, 2023-05-03:
The command works for me. Thx

02WYF23, 2024-04-21:
I solved this issue with conda.
sudo apt install ligblew-dev

clerdson, 2024-05-07:
c++ suports path include 
#include </Users/clerdson/Desktop/glew/include/GL/glew.h>