openai/mujoco-py #751: CentOs 7  distutils.errors.CompileError: command '/usr/bin/gcc' failed with exit code 1

lmx760581375, 2022-11-13:
Python 3.7.13 (default, Mar 29 2022, 02:18:16) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /data/home/mingxiaoli/anaconda3/envs/rl/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/root/mujoco_py/mujoco-py/mujoco_py -I/root/.mujoco/mujoco200/include -I/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/numpy/core/include -I/data/home/mingxiaoli/anaconda3/envs/rl/include/python3.7m -c /root/mujoco_py/mujoco-py/mujoco_py/cymj.c -o /root/mujoco_py/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.0_37_linuxcpuextensionbuilder/temp.linux-x86_64-cpython-37/root/mujoco_py/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /data/home/mingxiaoli/anaconda3/envs/rl/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/root/mujoco_py/mujoco-py/mujoco_py -I/root/.mujoco/mujoco200/include -I/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/numpy/core/include -I/data/home/mingxiaoli/anaconda3/envs/rl/include/python3.7m -c /root/mujoco_py/mujoco-py/mujoco_py/gl/osmesashim.c -o /root/mujoco_py/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.0_37_linuxcpuextensionbuilder/temp.linux-x86_64-cpython-37/root/mujoco_py/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
In file included from /root/mujoco_py/mujoco-py/mujoco_py/gl/osmesashim.c:1:
/usr/include/GL/osmesa.h:58:10: fatal error: GL/gl.h: No such file or directory
   58 | #include <GL/gl.h>
      |          ^~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/unixccompiler.py", line 186, in _compile
    self.spawn(compiler_so + cc_args + [src, '-o', obj] + extra_postargs)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/ccompiler.py", line 1007, in spawn
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/spawn.py", line 71, in spawn
    "command {!r} failed with exit code {}".format(cmd, exitcode)
distutils.errors.DistutilsExecError: command '/usr/bin/gcc' failed with exit code 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/mujoco_py/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/root/mujoco_py/mujoco-py/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/root/mujoco_py/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/root/mujoco_py/mujoco-py/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/root/mujoco_py/mujoco-py/mujoco_py/builder.py", line 272, in _build_impl
    so_file_path = super()._build_impl()
  File "/root/mujoco_py/mujoco-py/mujoco_py/builder.py", line 243, in _build_impl
    dist.run_commands()
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 968, in run_commands
    self.run_command(cmd)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
    cmd_obj.run()
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
    self.build_extensions()
  File "/root/mujoco_py/mujoco-py/mujoco_py/builder.py", line 143, in build_extensions
    build_ext.build_extensions(self)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 466, in build_extensions
    self._build_extensions_serial()
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 492, in _build_extensions_serial
    self.build_extension(ext)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 554, in build_extension
    depends=ext.depends,
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/ccompiler.py", line 599, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/data/home/mingxiaoli/anaconda3/envs/rl/lib/python3.7/site-packages/setuptools/_distutils/unixccompiler.py", line 188, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/bin/gcc' failed with exit code 1

lmx760581375, 2022-11-13:
i fixed it by method from issue 627，i hope the people in the back don't step in the hole again

epignatelli, 2023-01-11:
#627