openai/mujoco-py #753: Clang error when trying to compile mujoco-py

MateoMalidin, 2022-11-15:
**Describe the bug**
I am trying to run a program using the mujoco-py library. When I launch the program, a compilation error appears at the line `import mujoco_py`:

**Error Messages**
```
distutils.errors.CompileError: command '/usr/local/opt/llvm/bin/clang' failed with exit status 1
```

**Desktop (please complete the following information):**
 - OS: macos 12.1
 - Python Version 3.9.13
 - Mujoco Version 210
 - mujoco-py version 2.1
 - Compiler version : Clang 12.0.0
- Conda version : 22.9.0

**Additional context**
I followed the instructions about this problem available [here](https://github.com/openai/mujoco-py) :

```
brew install llvm
brew install boost
brew install hdf5

# Add this to your .bashrc/.zshrc:
export PATH="/usr/local/opt/llvm/bin:$PATH"

export CC="/usr/local/opt/llvm/bin/clang"
export CXX="/usr/local/opt/llvm/bin/clang++"
export CXX11="/usr/local/opt/llvm/bin/clang++"
export CXX14="/usr/local/opt/llvm/bin/clang++"
export CXX17="/usr/local/opt/llvm/bin/clang++"
export CXX1X="/usr/local/opt/llvm/bin/clang++"

export LDFLAGS="-L/usr/local/opt/llvm/lib"
export CPPFLAGS="-I/usr/local/opt/llvm/include"
```

But the error persists. I tried with different version of python (3.6, 3.9 ...), different version of mujoco-py (2.0.2.8, 2.0.2.13, 2.1 ...) but nothing works.

I also tried to do the same installation on ubuntu 22.04 and I get a similar error:

```
distutils.errors.CompileError: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
```

Would you know how to solve this problem ?

fatcatZF, 2023-02-06:
I have the same problem.

lzhhh93, 2023-03-03:
Same here. Any tipps?

mg19950, 2023-04-04:
@MateoMalidin Did you manage to resolve it?

happy-yan, 2024-02-04:
3.8 is ok