openai/mujoco-py #772: about import mujoco_py

zcswdt, 2023-06-12:
>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/usr/local/lib/python3.8/dist-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/usr/local/lib/python3.8/dist-packages/mujoco_py/builder.py", line 74, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", lib_path)
  File "/usr/local/lib/python3.8/dist-packages/mujoco_py/builder.py", line 120, in _ensure_set_env_var
    raise Exception("\nMissing path to your environment variable. \n"
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=/usr/local/lib/python3.8/dist-packages/torch/lib:/usr/local/lib/python3.8/dist-packages/torch_tensorrt/lib:/usr/local/cuda/compat/lib:/usr/local/nvidia/lib:/usr/local/nvidia/lib64
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/root/.mujoco/mujoco210/bin

abdulghani91, 2023-10-01:
I have the same problem I added the LD_LIBRARY_PATH, but it still does not work did you find any solution?

 import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/.local/lib/python3.10/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/user/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/user/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 74, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", lib_path)
  File "/home/user/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 120, in _ensure_set_env_var
    raise Exception("\nMissing path to your environment variable. \n"
Exception:
Missing path to your environment variable.
Current values LD_LIBRARY_PATH=/usr/local/cuda-12.2/lib64${LD_LIBRARY_PATH:+:}
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/user/.mujoco/mujoco210/bin

Anna-fence, 2024-01-16:
I meet the problem too. The solution in this issue https://github.com/openai/mujoco-py/issues/267 works for me.

sainavaneet, 2024-04-03:
**use this commands in terminal** 

`echo 'export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/user/.mujoco/mujoco210/bin' >> ~/.bashrc`
next
`source ~/.bashrc
`