openai/mujoco-py #773: Mujoco-py is incompatible with cpython

pseudo-rnd-thoughts, 2023-07-17:
**Describe the bug**
`Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'`

**To Reproduce**
Install cpython 3.0.0, import mujoco-py 

**Expected behavior**
mujoco-py compiles

**Error Messages**
From gymnasium's CI 
```
tests/wrappers/test_passive_env_checker.py:9: in <module>
    from tests.envs.test_envs import PASSIVE_CHECK_IGNORE_WARNING
<frozen importlib._bootstrap>:991: in _find_and_load
    ???
<frozen importlib._bootstrap>:975: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:671: in _load_unlocked
    ???
../lib/python3.8/site-packages/_pytest/assertion/rewrite.py:168: in exec_module
    exec(co, module.__dict__)
tests/envs/test_envs.py:10: in <module>
    from tests.envs.utils import (
tests/envs/utils.py:34: in <module>
    all_testing_initialised_envs: List[Optional[gym.Env]] = [
tests/envs/utils.py:35: in <listcomp>
    try_make_env(env_spec) for env_spec in gym.envs.registry.values()
tests/envs/utils.py:22: in try_make_env
    return env_spec.make(disable_env_checker=True).unwrapped
gymnasium/envs/registration.py:130: in make
    return make(self, **kwargs)
gymnasium/envs/registration.py:756: in make
    env_creator = load_env_creator(env_spec.entry_point)
gymnasium/envs/registration.py:545: in load_env_creator
    mod = importlib.import_module(mod_name)
../lib/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
gymnasium/envs/mujoco/__init__.py:1: in <module>
    from gymnasium.envs.mujoco.mujoco_env import MujocoEnv, MuJocoPyEnv  # isort:skip
gymnasium/envs/mujoco/mujoco_env.py:13: in <module>
    import mujoco_py
/mujoco-py/mujoco_py/__init__.py:2: in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
/mujoco-py/mujoco_py/builder.py:504: in <module>
    cymj = load_cython_ext(mujoco_path)
/mujoco-py/mujoco_py/builder.py:110: in load_cython_ext
    cext_so_path = builder.build()
/mujoco-py/mujoco_py/builder.py:226: in build
    built_so_file_path = self._build_impl()
/mujoco-py/mujoco_py/builder.py:278: in _build_impl
    so_file_path = super()._build_impl()
/mujoco-py/mujoco_py/builder.py:239: in _build_impl
    dist.ext_modules = cythonize([self.extension])
../lib/python3.8/site-packages/Cython/Build/Dependencies.py:1134: in cythonize
    cythonize_one(*args)
../lib/python3.8/site-packages/Cython/Build/Dependencies.py:1301: in cythonize_one
    raise CompileError(None, pyx_file)
E   Cython.Compiler.Errors.CompileError: /mujoco-py/mujoco_py/cymj.pyx
------------------------------- Captured stdout --------------------------------
Compiling /mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /mujoco-py/mujoco_py/cymj.pyx
------------------------------- Captured stderr --------------------------------

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_warning_callback
    global mju_user_warning
    py_warning_callback = warn
    mju_user_warning = c_warning_callback
                       ^
------------------------------------------------------------

/mujoco-py/mujoco_py/cymj.pyx:92:23: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_error_callback
    global mju_user_error
    py_error_callback = err_callback
    mju_user_error = c_error_callback
                     ^
------------------------------------------------------------

/mujoco-py/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'
```

JayanthRR, 2023-07-17:
Facing the same issue.

ChengDaHaI, 2023-07-18:
Same issue on my side

pseudo-rnd-thoughts, 2023-07-18:
I solved it with `pip install "cython<3"`

ezhang7423, 2023-07-21:
TYSM for figuring this out, wasted 3 hours today

solemnlyabsurd, 2023-07-21:
already have Cython==0.29.36 but the same error exists when pip install mujoco_py==2.0.2.7

qodfathr, 2023-07-21:
> already have Cython==0.29.36 but the same error exists when pip install mujoco_py==2.0.2.7

Same

ezhang7423, 2023-07-22:
I am using `mujoco-py==1.50.1.68`

qodfathr, 2023-07-22:
I cannot tell if this is directly related to cython 3 or not. I have 0.29.36 installed (and I've tried many versions from 0.27.2 on up) and the same error is occurring even on mujoco_py==2.0.2.13. I tried to determine if a patch to the underlying c++ compiler was at fault, but that does not seem to be the case.

pseudo-rnd-thoughts, 2023-07-22:
> I cannot tell if this is directly related to cython 3 or not. I have 0.29.36 installed (and I've tried many versions from 0.27.2 on up) and the same error is occurring even on mujoco_py==2.0.2.13. I [...]

I would be surprised as our CI should be failing if this was true but it is not failing with cython < 3 also if you are getting the same issue with `nogil` in the error then this is a `cython >= 3` thing

qodfathr, 2023-07-22:
> I would be surprised as our CI should be failing if this was true but it is not failing with cython < 3 also if you are getting the same issue with `nogil` in the error then this is a `cython >= 3` thing

I now have it working with `mujoco_py==2.0.2.5`, but no later `2.0.2.x` releases. I see that 2.0.2.5 is probably the last "official" 2.0.2 release, but [PyPi is misleading](https://pypi.org/project/mujoco-py/#history).

reginald-mclean, 2023-07-25:
I was able to use 
```python
pip install 'cython<3'
``` 
to compile mujoco-py 2.1.2.14 on Ubuntu 22.04, Python 3.8

Zebin-Li, 2023-07-25:
same, `cython<3` not works for me

zuhu2195, 2023-07-31:
I was facing the same issue in numpy-1.23.3 build with cython 3.0.0, using cython<3 worked for me.

HasarinduPerera, 2023-08-03:
> I solved it with `pip install "cython<3"`

Does not work for me, I am trying to install safety-gym.

mujoco200 
mujoco-py 2.1.2.14 
Ubuntu 20.04
Python 3.6

Anyone found any other ways to resolve?

TIA

zuhu2195, 2023-08-03:
What is your current Cython version?

HasarinduPerera, 2023-08-03:
> What is your current Cython version?

Cython     0.29.36

zuhu2195, 2023-08-03:
can you try updating your python to a higher version like 3.8.

HasarinduPerera, 2023-08-03:
> can you try updating your python to a higher version like 3.8.

No progress.

mynkpl1998, 2023-09-02:
Surprisingly, uninstalling Cython fixed the problem for me. 

Python: 3.8
gymnasium==0.29.1
mujoco==2.3.7

harryting-yiting, 2023-09-09:
1. Add a compiler directive " compiler_directives={'legacy_implicit_noexcept': True}" to the line 239 of mujoco_py/builder.py 

```cython
 def _build_impl(self):
        dist = Distribution({
            "script_name": None,
            "script_args": ["build_ext"]
        })
        dist.ext_modules = cythonize([self.extension],compiler_directives={'legacy_implicit_noexcept': True})
``` 
2. Update Cython to the latest 3.0.2 version.

```shell
pip install -U Cython
``` 

3. The reasons can be found here
https://github.com/scikit-learn/scikit-learn/issues/25609
https://github.com/cython/cython/issues/5637

zerlinwang, 2023-09-16:
> > > I would be surprised as our CI should be failing if this was true but it is not failing with cython < 3 also if you are getting the same issue with `nogil` in the error then this is a `cython >= 3` thing
> > 
> > 
> > I now have it working with `mujoco_py==2.0.2.5`, but no later `2.0.2.x` releases. I see that 2.0.2.5 is probably the last "official" 2.0.2 release, but [PyPi is misleading](https://pypi.org/project/mujoco-py/#history).
> 
> This Worked!

It works for my.

cangcn, 2023-09-19:
> > I solved it with `pip install "cython<3"`
> 
> Does not work for me, I am trying to install safety-gym.
> 
> mujoco200 mujoco-py 2.1.2.14 Ubuntu 20.04 Python 3.6
> 
> Anyone found any other ways to resolve?
> 
> TIA

mujoco-py 2.1.2.14  and mujoco200 are not match. Use mujoco_py 2.1.2.14 with mujoco210, or mujoco>2.0 <2.1 with mujoco200.

iLern, 2023-11-28:
> 1. Add a compiler directive " compiler_directives={'legacy_implicit_noexcept': True}" to the line 239 of **mujoco_py/builder.py**
> 
> ```cython
>  def _build_impl(self):
>         dist = Distribution({
>             "script_name": None,
>             "script_args": ["build_ext"]
>         })
>         dist.ext_modules = cythonize([self.extension],compiler_directives={'legacy_implicit_noexcept': True})
> ```
> 
> 2. Update Cython to the latest 3.0.2 version.
> 
> ```shell
> pip install -U Cython
> ```
> 
> 3. The reasons can be found here
>    [[MAINT, Cython] Implicit `noexcept` is deprecated in Cython 3.0 scikit-learn/scikit-learn#25609](https://github.com/scikit-learn/scikit-learn/issues/25609)
>    [[BUG] enum classes cannot be used in cdef class method declarations cython/cython#5637](https://github.com/cython/cython/issues/5637)

This works for my problem. Thanks a lot.

my env:
- Ubuntu 22.04
- python 3.8
- mujoco 2.1.0
- mujoco-py 2.1.2.14
- Cython 3.0.6

Zsk0, 2023-12-13:
> This works for my problem. Thanks a lot.
>
> my env:
>
> * Ubuntu 22.04
> * python 3.8
> * mujoco 2.1.0
> * mujoco-py 2.1.2.14
> * Cython 3.0.6

This works!
Thanks

YupuLu, 2024-01-07:
`pip install Cython==3.0.0a10` works for me.

My env:
Ubuntu 20.04
python 3.8.17
mujoco 2.1.0
mujoco-py 2.1.2.14

dhruvmsheth, 2024-06-28:
Here are the steps that worked for me:

```shell
pip install "cython<3"
conda install gxx_linux-64
sudo ln -sf /usr/bin/gcc /home/dhruvstra/anaconda3/envs/argus/bin/x86_64-conda-linux-gnu-cc
```

The `/home/dhruvstra/anaconda3/envs/argus/bin/x86_64-conda-linux-gnu-cc` should be your path to where conda linux gnu is installed. It might be located at `usr/bin/`

and the most important one was:
`conda install -c conda-forge gcc=12.1.0`

gaisibo, 2024-07-26:
> Here are the steps that worked for me:
>
> ```shell
> pip install "cython<3"
> conda install gxx_linux-64
> sudo ln -sf /usr/bin/gcc /home/dhruvstra/anaconda3/envs/argus/bin/x86_64-conda-linux-gnu-cc
> ``` [...]

This works for me! Thank you!

awesomeNabi, 2024-09-25:
> Here are the steps that worked for me:
>
> ```shell
> pip install "cython<3"
> conda install gxx_linux-64
> sudo ln -sf /usr/bin/gcc /home/dhruvstra/anaconda3/envs/argus/bin/x86_64-conda-linux-gnu-cc
> ``` [...]

wonderful!!!it works

varungupta31, 2024-12-03:
None of the other approaches worked for me, I started afresh and followed this:

https://github.com/openai/mujoco-py/issues/803

This finally worked. If you just want to run `mujoco`, this is worth a shot :)

Copy-pasting the contents of the linked discussion, credit to @Josh00-Lu

```
## Create conda
conda create -n demo python=3.9 mesalib glew glfw -c conda-forge -y
conda activate demo

## Install mujoco dependency
USER_DIR=$USER
echo $USER_DIR
wget -c "https://mujoco.org/download/mujoco210-linux-x86_64.tar.gz"
mkdir -p /home/$USER_DIR/.mujoco
cp mujoco210-linux-x86_64.tar.gz /home/$USER_DIR/mujoco.tar.gz
rm mujoco210-linux-x86_64.tar.gz
mkdir -p /home/$USER_DIR/.mujoco
tar -zxvf /home/$USER_DIR/mujoco.tar.gz -C /home/$USER_DIR/.mujoco
echo "export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/$USER_DIR/.mujoco/mujoco210/bin" >> ~/.bashrc
echo "export MUJOCO_PY_MUJOCO_PATH=/home/$USER_DIR/.mujoco/mujoco210" >> ~/.bashrc

# activate 
source ~/.bashrc
conda activate demo

#  install 
export CC="/usr/bin/gcc"
pip install "mujoco_py>=2.0"
pip install "cython<3"

# compile mujoco using gcc
python
import mujoco_py
```