openai/mujoco-py #776: Installation problem( Mujoco-py 1.50.1.5 )  ModuleNotFoundError: No module named 'Cython'

neuronphysics, 2023-07-30:
Hi,

 I am trying to test a baseline model which is published in 2019 and the code is written in tensorflow 1.

I'm looking for some help with its installation of Mujoco-py on google colab.

Here is my approach for installing mujoco-py :

```
! wget https://repo.anaconda.com/miniconda/Miniconda3-py37_4.12.0-Linux-x86_64.sh
! chmod +x Miniconda3-py37_4.12.0-Linux-x86_64.sh
! bash ./Miniconda3-py37_4.12.0-Linux-x86_64.sh -b -f -p /usr/local/
!conda update conda -y

!pip install --upgrade pip
!pip install --upgrade setuptools
!pip install wheel

!pip install Cython
sys.path.append('/usr/local/lib/python3.7/site-packages')
!python --version
!apt-get -qq update
!apt-get -qq install -y libosmesa6-dev libgl1-mesa-glx libglfw3 libgl1-mesa-dev libglew-dev patchelf libxml2-dev libxslt1-dev 
!apt-get -qq install -y gcc libpython3.7-dev build-essential 
!apt-get -qq install -y libssl-dev zlib1g-dev software-properties-common python3-pip libgmp-dev portaudio19-dev libffi-dev
!apt-get -qq install -y libblas-dev libatlas-base-dev xvfb xorg-dev libsdl2-dev swig cmake libc-dev
os.environ['PYTHONPATH']+=':/usr/local:/usr/local/lib/python3.7/site-packages'
# Get Mujoco
import shutil

try:
    folder_path = '/root/.mujoco'
    shutil.rmtree(folder_path)
    print('Folder and its content removed')
except:
    print('Folder not deleted')
!mkdir ~/.mujoco
!mkdir /root/.mujoco/mujoco150
os.chdir('/root/.mujoco')
!wget -q https://files.pythonhosted.org/packages/25/72/697c6029ec3e7e0682f35f3758f3fb5d3960795ff22b277383146f42aa00/mujoco-py-1.50.1.5.tar.gz -O mujoco.tar.gz
!tar -zxf mujoco.tar.gz -C "$HOME/.mujoco"
source_dir = '/root/.mujoco/mujoco-py-1.50.1.5'
target_dir = '/root/.mujoco/mujoco150'
    
file_names = os.listdir(source_dir)
    
for file_name in file_names:
    shutil.move(os.path.join(source_dir, file_name), target_dir)

os.chdir('/root/.mujoco/mujoco150')
!ls -ltr
#!conda install pip
!pip install -r requirements.txt
!pip install -r requirements.dev.txt

!python setup.py build_ext --inplace
!pip install lxml
!pip install pymssql
!pip3 install -e . --no-cache

#!pip install mujoco-py==1.50.1.56
print("after installing....")
!ls -ltr /usr/bin/x86_64-linux-gnu-gcc
!ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so
# Add it to the actively loaded path and the bashrc path (these only do so much)
!echo 'export LD_PRELOAD=$LD_PRELOAD:/usr/lib/x86_64-linux-gnu/libGLEW.so' >> "/root/.bashrc"

# THE ANNOYING ONE, FORCE IT INTO LDCONFIG SO WE ACTUALLY GET ACCESS TO IT THIS SESSION
!echo "/root/.mujoco/mujoco150/bin" > /etc/ld.so.conf.d/mujoco_ld_lib_path.conf
!ldconfig

!cp /content/mjkey.txt /root/.mujoco
!cp /content/mjkey.txt ~/.mujoco/mujoco150/bin
!source /root/.bashrc
!pip install llvmlite
_mujoco_run_once = False
os.environ['LD_LIBRARY_PATH']='/usr/lib64-nvidia:/root/.mujoco/mujoco150/bin:/usr/lib/nvidia:/usr/local/nvidia/lib:/usr/local/nvidia/lib64'
os.environ['LD_LIBRARY_PATH']=os.environ['LD_LIBRARY_PATH']+':/root/.mujoco/mjpro150/bin'
os.environ['LD_PRELOAD']='/usr/lib/x86_64-linux-gnu/libGLEW.so'
!export CFLAGS="-Wno-narrowing"
if not _mujoco_run_once:
  # Add it to the actively loaded path and the bashrc path (these only do so much)
  try:
    os.environ['LD_LIBRARY_PATH']=os.environ['LD_LIBRARY_PATH'] + ':/root/.mujoco/mujoco150/bin:/usr/lib/nvidia:/usr/local/nvidia/lib:/usr/local/nvidia/lib64'
    os.environ['PATH']=os.environ['LD_LIBRARY_PATH']+os.environ['PATH']
  except KeyError:
    os.environ['LD_LIBRARY_PATH']='/root/.mujoco/mujoco150/bin'
    os.environ['PATH']=os.environ['LD_LIBRARY_PATH']+os.environ['PATH']
  try:
    os.environ['LD_PRELOAD']=os.environ['LD_PRELOAD'] + ':/usr/lib/x86_64-linux-gnu/libGLEW.so'
  except KeyError:
    os.environ['LD_PRELOAD']='/usr/lib/x86_64-linux-gnu/libGLEW.so'
  # presetup so we don't see output on first env initialization
  os.environ['MUJOCO_PY_MUJOCO_PATH'] = '/root/.mujoco/mujoco150'
  os.environ['MUJOCO_PY_MJKEY_PATH'] = '/root/.mujoco/mjkey.txt'
  
  print("pythonpath......")

  !echo $PYTHONPATH
  sys.path.insert(1, "/root/.mujoco/")
  if "/usr/lib/python3.10" in sys.path:
     sys.path.remove('/usr/lib/python3.10')

  if "/usr/local/lib/python3.10/dist-packages" in sys.path:
     sys.path.remove('/usr/local/lib/python3.10/dist-packages')

  if "/usr/include/python3.10" in sys.path:
     sys.path.remove('/usr/include/python3.10')

  print(sys.path)
  !python --version
  
  import mujoco_py
  mj_path, _ = mujoco_py.utils.discover_mujoco()
  xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
  model = mujoco_py.load_model_from_path(xml_path)
  sim = mujoco_py.MjSim(model)

  print(sim.data.qpos)
  _mujoco_run_once = True
```

Here is my error message which I can not get rid off:
```
Ppythonpath......
/env/python:/usr/local:/usr/local/lib/python3.7/site-packages
['/content', '/env/python', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '', '/usr/local/lib/python3.10/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.10/dist-packages/IPython/extensions', '/root/.ipython', '/usr/local/lib/python3.7/site-packages']
['/content', '/root/.mujoco/', '/env/python', '/usr/lib/python310.zip', '/usr/lib/python3.10/lib-dynload', '', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.10/dist-packages/IPython/extensions', '/root/.ipython', '/usr/local/lib/python3.7/site-packages']
Python 3.7.13
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
[<ipython-input-5-3af954154eab>](https://localhost:8080/#) in <cell line: 87>()
    120   get_ipython().system('python --version')
    121 
--> 122   import mujoco_py
    123   mj_path, _ = mujoco_py.utils.discover_mujoco()
    124   xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')

1 frames
[~/.mujoco/mujoco150/mujoco_py/__init__.py](https://localhost:8080/#) in <module>
----> 1 from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
      2 from mujoco_py.generated import const
      3 from mujoco_py.mjviewer import MjViewer, MjViewerBasic
      4 from mujoco_py.version import __version__, get_version
      5 

[~/.mujoco/mujoco150/mujoco_py/builder.py](https://localhost:8080/#) in <module>
     12 
     13 import numpy as np
---> 14 from Cython.Build import cythonize
     15 from Cython.Distutils.old_build_ext import old_build_ext as build_ext
     16 

ModuleNotFoundError: No module named 'Cython'

---------------------------------------------------------------------------
NOTE: If your import is failing due to a missing package, you can
manually install dependencies using either !pip or !apt.

To view examples of installing some common dependencies, click the
"Open Examples" button below.
---------------------------------------------------------------------------
```
I will be really grateful if someone can suggest a way to solve this error. 🙏🏻

Enigma-cit, 2023-10-22:
@neuronphysics I also encountered the same problem,
could you rectify it, if so pls pls share the solution.

sainavaneet, 2024-05-08:
try installing 
 `pip install 'Cython<3'`