openai/mujoco-py #777: mujoco1.50 - Could not find GCC executable.

the-jasonl, 2023-08-02:
**Describe the bug**
I want to run `pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'` on macOS but it returns

```
File "/private/var/folders/1t/t0dzx5fn3jn1r9lqx_2j3m9m0000gn/T/pip-install-i8wi8lki/mujoco-py_48d744c3b7c64c6abb74252cb291ab88/mujoco_py/builder.py", line 320, in _build_impl
      raise RuntimeError(
  RuntimeError: Could not find GCC executable.
  
  HINT: On OS X, install GCC with `brew install gcc`.
```

I have the following gcc version installed:

```
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin22.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

and am running Python version:

```
Python 3.9.17 (main, Jun 15 2023, 08:01:12) 
[Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
```

What is the problem here?

The ultimate goal is to install and run `gym[mujoco]==0.19`.

**To Reproduce**
Use setup above and run `pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'` on macOS

**Expected behavior**
Successful install

**Error Messages**
See above

**Desktop (please complete the following information):**
 - OS: macOS 13.4.1
 - Python Version: 3.9.17
 - Mujoco Version 1.50
 - mujoco-py version: trying to install

lukaskocian, 2023-09-05:
Hello Jason,
have you ended up fixing the problem? I got totally the same and can't get my head around it.

maxjkwang, 2024-05-30:
I think the reason for this is in `builder.py`, it only looks for GCC versions older than (and including) gcc-9. I tried to add my GCC path to the `c_compilers` list, but it would cause other compile errors, so I've resorted to using docker.
I'm not sure if future GCC versions are not supported at all, but if anyone wants to try this feel free to do so.