openai/mujoco-py #780: installation error on m1 mac Sonoma

qiuruiyu, 2023-11-08:
**Describe the bug**
When I tried to use [(#682)](https://github.com/openai/mujoco-py/issues/682) to install mujoco_py on my mac, error like: 
```bash
ImportError: dlopen(/Users/joseph/miniforge3/envs/ppo/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so, 0x0002): Library not loaded: @rpath/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib
Referenced from: <34A1F6DB-46E9-36E0-AD4B-48B493C455DD> /Users/joseph/miniforge3/envs/ppo/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so
```
I'm not sure whether it's because of my macos version, I have tried lots of ways to solve it but didn't succeed. 

**To Reproduce**
with macos 14, mujoco 2.1.1, mujoco_py 2.1.12.14, then run script in [(#682)](https://github.com/openai/mujoco-py/issues/682) 

**Expected behavior**
successfully install mujoco_py and run with gymnasium 

**Error Messages**
ImportError: dlopen(/Users/joseph/miniforge3/envs/ppo/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so, 0x0002): Library not loaded: @rpath/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib
  Referenced from: <34A1F6DB-46E9-36E0-AD4B-48B493C455DD> /Users/joseph/miniforge3/envs/ppo/lib/python3.8/site-packages/mujoco_py/generated/cymj_2.1.2.14_38_macextensionbuilder_38.so
  Reason: tried: '/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/.mujoco/mujoco210/bin/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/joseph/.mujoco/mujoco210/bin/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/miniforge3/envs/ppo/lib/python3.8/site-packages/mujoco_py/generated/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/aarch64-apple-darwin23/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/aarch64-apple-darwin23/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/.mujoco/mujoco210/bin/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/joseph/.mujoco/mujoco210/bin/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/miniforge3/envs/ppo/lib/python3.8/site-packages/mujoco_py/generated/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/aarch64-apple-darwin23/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/aarch64-apple-darwin23/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/gcc/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Cellar/gcc@11/11.4.0/lib/gcc/11/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/miniforge3/envs/ppo/bin/../lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file), '/Users/joseph/miniforge3/envs/ppo/bin/../lib/MuJoCo.framework/Versions/A/libmujoco.2.1.1.dylib' (no such file)

**Desktop (please complete the following information):**
 - OS: [macOS 14.0]
 - Python Version [3.8.18] —— (conda, miniforge3, arm64)
 - Mujoco Version [2.1.1]
 - mujoco-py version [2.1.12.14]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: ' ' (empty)
 - output of: `echo $HOME`: '/Users/joseph'
 - output of: `echo $USER`: joseph

**Additional context**
No

joanvelja, 2024-01-24:
Did you end up solving this? I have been bumping my head onto this the entire day....

qiuruiyu, 2024-01-24:
您好，感谢您的来信。已确认收到您的邮件，我会尽快处理您的邮件，谢谢。Hello！This is an automatic reply confirming that your email was received. Your email will be handled as soon as possible. Thank you！                                                      仇睿瑜                                              Joseph QIUThis is an automatic reply, confirming that your e-mail was received.Thank you

qiuruiyu, 2024-01-24:
> Did you end up solving this? I have been bumping my head onto this the entire day....

Yes, after I carefully read the error, all of these direct to one is that, xxxx (no such file). It has been a long time since I solved the problem, I am not sure it is because the file name is not correct or the file is not exist. No matter how, you just need to rename or copy the file from somewhere else to the destination: 
`/Users/joseph/miniforge3/envs/ppo/lib/MuJoCo.framework/Versions/A/xxxxx.dylib`

Chenrf1121, 2024-04-21:
Did you solve this problem?