openai/mujoco-py #785: after installing mujoco_py I get this cython error

davidireland3, 2024-01-18:
I'm installing on Linux and get as far as being able to import mujoco_py in python until I get the following error each time. Does anyone know what is causing this? 

```
Compiling /home/USER/.mujoco/mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/USER/.mujoco/mujoco-py/mujoco_py/cymj.pyx
performance hint: /home/USER/.mujoco/mujoco-py/mujoco_py/cymj.pyx:67:5: Exception check on 'c_warning_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_warning_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_warning_callback' to allow an error code to be returned.
performance hint: /home/USER/.mujoco/mujoco-py/mujoco_py/cymj.pyx:104:5: Exception check on 'c_error_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_error_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_error_callback' to allow an error code to be returned.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_warning_callback
    global mju_user_warning
    py_warning_callback = warn
    mju_user_warning = c_warning_callback
                       ^
------------------------------------------------------------
```

Wcy0802, 2024-01-20:
i have same issue, wait someone can solve

akrv, 2024-01-23:
I downgraded the Cython version `pip install cython<3.0` that fixed this error. but I have other errors now.

closed 2024-01-23 by davidireland3

whosthereme, 2024-01-29:
anyone fixed it? 
i have same issue.
i tried the solution they provided but never work.