openai/mujoco-py #786: Cython.Compiler.Errors.CompileError: D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\cymj.pyx

Wcy0802, 2024-01-20:
**Describe the bug**

>>> import mujoco_py
Compiling D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\cymj.pyx because it changed.
[1/1] Cythonizing D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\cymj.pyx
warning: D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\generated/wrappers.pxi:4:0: Dotted filenames ('../pxd/mujoco.pxd') are deprecated. Please use the normal Python package directory layout.

Error compiling Cython file:
------------------------------------------------------------
...
#   with wrap_mujoco_warning():
#       mj_somefunc()
cdef object py_warning_exception = None

cdef void c_warning_callback(const char *msg) with gil noexcept:
                                                      ^
------------------------------------------------------------

D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\cymj.pyx:65:55: Syntax error in C variable declaration
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\builder.py", line 503, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\builder.py", line 233, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\Cython\Build\Dependencies.py", line 1117, in cythonize
    cythonize_one(*args)
  File "D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\Cython\Build\Dependencies.py", line 1284, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: D:\Anaconda\Anaconda3\envs\py310torch2.1\lib\site-packages\mujoco_py-2.0.2.1-py3.10.egg\mujoco_py\cymj.pyxhavior.

Anyone can solve this issue?

sainavaneet, 2024-04-03:
i am also having same issue

Luckykantnayak, 2024-06-10:
Any resolve yet ?

sainavaneet, 2024-06-11:
> Any resolve yet ?

try doing this command `pip install Cython<3`

coollingomg, 2024-10-24:
try doing this command pip install Cython==3.0.0a10

Shreyojit, 2025-03-05:
> try doing this command pip install Cython==3.0.0a10

This works thanks man.