openai/mujoco-py #787: mujoco-py is not available now?

jiho314, 2024-02-03:
I tried everything but below code brings various errors.
Is mujoco-py unavailable now due to the new mujoco 3.0?

```
from mujoco_py import load_model_from_path
```

error1 : fixed by downgrading cython
```
Cython.Compiler.Errors.CompileError: /home/qkrwlgh0314/miniconda3/envs/uhc-new/lib/python3.8/site-packages/mujoco_py/cymj.pyx
```

error2:  
 i've tried various gcc versions but failed to fix the error
```
    spawn(cmd, dry_run=self.dry_run, **kwargs)
  File "/home/qkrwlgh0314/miniconda3/envs/uhc2/lib/python3.8/site-packages/setuptools/_distutils/spawn.py", line 70, in spawn
    raise DistutilsExecError(
distutils.errors.DistutilsExecError: command '/opt/ohpc/pub/compiler/gcc/8.3.0/bin/gcc' failed with exit code 1

```

Hoeng317, 2024-03-14:
What cython version did you downgrade to?

akharedeepak, 2024-07-07:
@jiho314 Did you solve it? 

I am facing the same issue.

jiho314, 2024-07-08:
I somehow fixed the error by following the install guidance from each git repository. 
I remember that 'pip install "cython<3”' and adding environment variables was necessary.