openai/mujoco-py #795: I am unable to import mujoco_py

sainavaneet, 2024-04-03:
**Error Messages**

Python 3.8.10 (default, Nov 22 2023, 10:22:35) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling /home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/cymj.pyx
performance hint: .local/lib/python3.8/site-packages/mujoco_py/cymj.pyx:67:5: Exception check on 'c_warning_callback' will always require the GIL to be acquired.
Possible solutions:
        1. Declare 'c_warning_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
        2. Use an 'int' return type on 'c_warning_callback' to allow an error code to be returned.
performance hint: .local/lib/python3.8/site-packages/mujoco_py/cymj.pyx:104:5: Exception check on 'c_error_callback' will always require the GIL to be acquired.
Possible solutions:
        1. Declare 'c_error_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
        2. Use an 'int' return type on 'c_error_callback' to allow an error code to be returned.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_warning_callback
    global mju_user_warning
    py_warning_callback = warn
    mju_user_warning = c_warning_callback
                       ^
------------------------------------------------------------

.local/lib/python3.8/site-packages/mujoco_py/cymj.pyx:92:23: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to the type of 'c_warning_callback'.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_error_callback
    global mju_user_error
    py_error_callback = err_callback
    mju_user_error = c_error_callback
                     ^
------------------------------------------------------------

.local/lib/python3.8/site-packages/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to the type of 'c_error_callback'.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 296, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/builder.py", line 239, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
    cythonize_one(*args)
  File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: /home/navaneet/.local/lib/python3.8/site-packages/mujoco_py/cymj.pyx

**can someone help me**

closed 2024-04-07 by sainavaneet

ZimaBlue307, 2024-04-11:
I encounted the same problem, how did you solved it?

skr3178, 2024-07-01:
https://github.com/openai/mujoco-py/issues/291#issuecomment-1738943394