openai/mujoco-py #798: Issue when imoporting mujoco_py (crypt.h: No such file or directory)

jayaramreddy10, 2024-06-14:
**Describe the bug**
Hi, I installed mujoco_py using pip (pip3 install -U 'mujoco-py<2.2,>=2.1') but when trying to import, I am getting error message.

**Error Messages**
Including more/longer error messages gives us more information to work with.
  >>> import mujoco_py
  In file included from /home/jayaram/.mujoco/mujoco-py/mujoco_py/cymj.c:57:
  /home/jayaram/miniconda3/envs/psl2/include/python3.8/Python.h:44:10: fatal error: crypt.h: No such file or directory
     44 | #include <crypt.h>
        |          ^~~~~~~~~
  compilation terminated.
  Traceback (most recent call last):
    File "/home/jayaram/miniconda3/envs/psl2/lib/python3.8/site-packages/setuptools/_distutils/unixccompiler.py", line 188, in _compile
      self.spawn(compiler_so + cc_args + [src, '-o', obj] + extra_postargs)
    File "/home/jayaram/miniconda3/envs/psl2/lib/python3.8/site-packages/setuptools/_distutils/ccompiler.py", line 1041, in spawn
      spawn(cmd, dry_run=self.dry_run, **kwargs)
    File "/home/jayaram/miniconda3/envs/psl2/lib/python3.8/site-packages/setuptools/_distutils/spawn.py", line 68, in spawn
      raise DistutilsExecError(f"command {cmd!r} failed with exit code {exitcode}")
  distutils.errors.DistutilsExecError: command '/home/jayaram/miniconda3/envs/psl2/bin/gcc' failed with exit code 1

**Desktop (please complete the following information):**
 - OS: [Ubuntu 22.04]
 - Python Version [e.g. 3.8]
 - Mujoco Version [e.g. 2.3.5]

Did I miss anything? Please let me know if any other details are required?

Yushi-Du, 2025-01-04:
Facing the same issue, have you solved it now? Thank you very much.

alexscavo, 2025-01-04:
Same here, I didn't find anywhere a valid solution