openai/mujoco-py #803: [Tutorial]: Tested solution for install mujoco_py on current linux server without root, Enjoy

Josh00-Lu, 2024-08-19:
Recommendation: 
Ubuntu, Conda

```bash
## Create conda
conda create -n demo python=3.9 mesalib glew glfw -c conda-forge -y
conda activate demo

## Install mujoco dependency
USER_DIR=demo
echo $USER_DIR
wget -c "https://mujoco.org/download/mujoco210-linux-x86_64.tar.gz"
mkdir -p /home/$USER_DIR/.mujoco
cp mujoco210-linux-x86_64.tar.gz /home/$USER_DIR/mujoco.tar.gz
rm mujoco210-linux-x86_64.tar.gz
mkdir -p /home/$USER_DIR/.mujoco
tar -zxvf /home/$USER_DIR/mujoco.tar.gz -C /home/$USER_DIR/.mujoco
echo "export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/$USER_DIR/.mujoco/mujoco210/bin" >> ~/.bashrc
echo "export MUJOCO_PY_MUJOCO_PATH=/home/$USER_DIR/.mujoco/mujoco210" >> ~/.bashrc

# activate 
source ~/.bashrc
conda activate demo

#  install 
export CC="/usr/bin/gcc"
pip install "mujoco_py>=2.0"
pip install "cython<3"

# compile mujoco using gcc
python
import mujoco_py
```

Tested: Ubuntu 22.04.4 LTS

Good luck, enjoy.

varungupta31, 2024-12-03:
@Josh00-Lu 
Thank you so much! 
Finally able to run mujoco.

<3

shadielfares, 2025-02-08:
YOU ARE THE BEST HUMAN BEING TO EVER EXIST EVER, was in loops trying to get this done.

Worked on Ubuntu 24.04, with conda.

As it stands the current [SpinningUp OpenAI](https://spinningup.openai.com/en/latest/user/installation.html) docs are deprecated.

Thank you.