openai/mujoco-py #807: mujoco_py looking for the wrong mujoco installation

NoahRCarver, 2024-10-23:
I have two installations of mujoco for two different projects
2.3.1 is installed in ~/ml4kp_ws/mujoco-2.3.1/bin
2.1.0 is installed in ~/.mujoco/mujoco210/bin and is the one i want my to be found at the moment.

[image]

tlaurie99, 2024-10-31:
@NoahRCarver 
I had this issue and it was a matter of opening the ~/.bashrc file and putting in:
```
export MUJOCO_PY_MUJOCO_PATH="$HOME/.mujoco/mujoco210"
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$MUJOCO_PY_MUJOCO_PATH/bin
```
then running in a terminal:
```
source ~/.bashrc
```

One part I got hung up on is exiting the venv, activating a new one, and then trying:
```
python3
import mujoco_py
```
so that the env. variables get refreshed. Hope it helps!

carloskiyoto, 2025-01-24:
I'm facing the same inssue and didn't work, I'm guessing is my python version
which version are you using?